Mr-HaleYa/MWS-Hardware #10: image/name links

vlost, 2020-06-15:
when you click on your image/name none of the links work

Mr-HaleYa, 2020-06-15:
@vlost can you explain with more than 12 words? what page? user or admin? screenshot?

0xCTF, 2020-06-15:
yes i didn't worked on that dropdown list yet, i just was about to delete it and put only logout
do we need those options?
i was thinking about settings, where the account owner can change his details(not username) and password without needing to search for it in manage admin and user will be able to change his details too
for activity log, do we need it ?
idk about profile, it will show details about the account owner

Mr-HaleYa, 2020-06-15:
@0xCTF I think we should only have settings and log out and in the settings, they can change their email, password, phone number, Billing info, and address.

0xCTF, 2020-06-16:
**Fixed**
[image]
Note: still settings page to work on it

0xCTF, 2020-06-16:
**settings works now**

Mr-HaleYa, 2020-06-17:
Great work it is working

closed 2020-06-17 by Mr-HaleYa