Mr-HaleYa/MWS-Hardware #7: click on "more details" in list of stations

vlost, 2020-06-15:
on station list page get rid of the "More Details" and just click anywhere in the box.
...but I really want a sort able table, if it's not too hard,  Your boxes are cool but when a customer has 100 stations it will be easier to sort thought them with a table

Mr-HaleYa, 2020-06-15:
We need to discuss how to do this properly before attempting.

0xCTF, 2020-06-15:
i just mentionned that in https://github.com/Mr-HaleYa/mywatersitebuglog/issues/9
that's what you mean right ?
the table structure will mess up the look of the stations cards(the rectangle that have the station)
so my idea is a search bar and a pages number in like the manage admin/user

0xCTF, 2020-06-17:
**"on station list page get rid of the "More Details" and just click anywhere in the box."**
Done

Mr-HaleYa, 2020-06-17:
only working for admins

Mr-HaleYa, 2020-06-17:
fixed for all, great job

closed 2020-06-17 by Mr-HaleYa