Mr-HaleYa/MWS-Hardware #17: Settings password autofilled cause password changing

0xCTF, 2020-06-17:
When there's a registered password in the browser for a username for example admin and you open an account with username admin and open settings window and you change some details but not the password, and if your password is not the one is regestered in the browser, so it will change to that registered one because the password field filled automatically

Mr-HaleYa, 2020-06-17:
Yes, I kept having that problem when creating new users it would give them my username and password because it would autofill it into the tabs and I would have to delete it out and then type it in manually and it was really annoying. To remove autofill from the entire website except for the homepage so that you can autofill your username and password into the login

0xCTF, 2020-06-18:
> Yes, I kept having that problem when creating new users it would give them my username and password because it would autofill it into the tabs and I would have to delete it out and then type it in [...]

That's different problem, idk if it's possible to disable autofill for all the pages, i'll see how i can do it after finishing higher priority bugs

**The issues i posted should be fixed 
check the video below to understand more, it's about changing user/admin password without noticing ut's autofilled and ajax "Confirm password" doesn't work in that case, only when you click on the filed then click out of it**
https://drive.google.com/file/d/17ulgNi83A2TsSCgjkTp-M-d-Z5cZsptv/view?usp=sharing
_sorry for the music_

Mr-HaleYa, 2020-06-18:
@0xCTF 
Is the video you linked to before or after you "fixed" it?

0xCTF, 2020-06-18:
> @0xCTF
> Is the video you linked to before or after you "fixed" it?

after i fixed it, so now, if user have a value in the password field(from browser autofill), and that value is old password, and he didn't noticed it which is what happend to me and i kicked out from my account, so he/she will get the field "confirm password" required

Mr-HaleYa, 2020-06-18:
lol im just going to say yes and close this 😄

closed 2020-06-18 by Mr-HaleYa