Mr-HaleYa/MWS-Hardware #18: Add least reading timestamp

Mr-HaleYa, 2020-06-17:
should show the timestamp of the last reading that that was taken, have it as hh-mm dd-MM

[image]

0xCTF, 2020-06-18:
timestamp is now formated as hh-mm dd-MM
**Fixed**

Mr-HaleYa, 2020-06-18:
Can you make that into 12 hours since people in America don't you used 24 hours, also, the timing is wrong because we save the data in UTC so there needs to be some kind of function that converts it into our time zone

0xCTF, 2020-06-18:
**Can you make that into 12 hours since people in America don't you used 24 hours,**
Done
the timing is wrong because we save the data in UTC so there needs to be some kind of function that converts it into our time zone. That goes to all things that use dates, like the graphs
MST ?

0xCTF, 2020-06-19:
**Fixed**
now it check ip adress location timezone and convert UTC to that timezone

0xCTF, 2020-06-19:
Note: the code somehow is not stable, sometimes it gives the wrong timezone 
check from here
https://www.mywater.site/time.php

0xCTF, 2020-06-19:
the other solution is making a new field in Manage > USER to add the new client timezone and save it in database
that better than ip adress, from which it said here https://stackoverflow.com/questions/743505/how-to-get-time-zone-through-ip-address-in-php

but if the client is in travel in different country/state, things will not work correclty only if he/she changed timezone from Settings(the top right option)

0xCTF, 2020-06-19:
nevermind, the code is not correct; still working on it.

Mr-HaleYa, 2020-06-20:
> Note: the code somehow is not stable, sometimes it gives the wrong timezone
> check from here
> https://www.mywater.site/time.php

it is working for ma and always gives me the same TZ. If you use a VPN it will switch to all over the world

Mr-HaleYa, 2020-06-20:
@0xCTF I think we should just add it as a field for users during signup and add it to the user table as a new column. they should not be traveling to other timezones and if they do they can do the math.

0xCTF, 2020-06-20:
**All Done**

Mr-HaleYa, 2020-06-21:
Its looking good. The only thing that is missing is daylight savings time but that shouldn't be to hard to do. opened on #31

closed 2020-06-21 by Mr-HaleYa