Mr-HaleYa/MWS-Hardware #19: cant go back to "water highlighted" style on map once clicked to a different style

vlost, 2020-06-18:


Mr-HaleYa, 2020-06-18:
@vlost you need to have descriptions and/or a screenshot or vid

Mr-HaleYa, 2020-06-18:
https://drive.google.com/file/d/1338xpq5ALRVyvhpO0id5bs54PmE8s72z/view

0xCTF, 2020-06-18:
i was trying for couple hours, still can't do it and i don't think it's possible.
Here's two answers from stackoverflow
https://stackoverflow.com/questions/3918608/google-maps-api-styled-map-with-satellite-water
[image]

https://stackoverflow.com/questions/46279135/google-styled-maps-can-be-applied-to-satellite-map-type
[image]

Mr-HaleYa, 2020-06-18:
So if you switch a map and then switch back it just deletes your styling?

0xCTF, 2020-06-18:
> So if you switch a map and then switch back it just deletes your styling?

Yeah because it back to the real map instead of the custom one, i'll ask stackoverflow about that

closed 2020-06-19 by Mr-HaleYa