Mr-HaleYa/MWS-Hardware #20: User changing station name crashes website

vlost, 2020-06-18:
[image]

Mr-HaleYa, 2020-06-18:
Notice: Undefined index: flowonOroff in /home/u287141768/domains/mywater.site/public_html/process_user.php on line 30

Notice: Undefined index: onOroff in /home/u287141768/domains/mywater.site/public_html/process_user.php on line 31

Fatal error: Uncaught PDOException: SQLSTATE[42S22]: Column not found: 1054 Unknown column 'show_flow' in 'field list' in /home/u287141768/domains/mywater.site/public_html/process_user.php:43 Stack trace: #0 /home/u287141768/domains/mywater.site/public_html/process_user.php(43): PDOStatement->execute(Array) #1 {main} thrown in /home/u287141768/domains/mywater.site/public_html/process_user.php on line 43

0xCTF, 2020-06-18:
**Fixed**

closed 2020-06-18 by Mr-HaleYa