Mr-HaleYa/MWS-Hardware #40: way to many decimals

Mr-HaleYa, 2020-06-24:
When converting to a different unit type there is no cut-off so some of them have like 30 decimal places. Have it cut off at the hundredths place (0.00). This will have to be added to all unit conversions 

https://drive.google.com/file/d/1vSc_FxG7q76tW5cKdd1pECdn2AcsHaOW/view

Mr-HaleYa, 2020-06-25:
this is what I use to get 2 decimals
``` $flow=number_format($flowRate, 2, '.', ""); //2 decimal points ```
it corrects rounding errors so it does the math right

0xCTF, 2020-06-26:
ok but many values in some units will be 0.00 because the full value is like 0.0000654597
is that ok ?

Mr-HaleYa, 2020-06-26:
temp is the only one that needs to be forced .00 the others are fine .0000000000000

0xCTF, 2020-06-26:
**Fixed**

Mr-HaleYa, 2020-06-27:
I like it

closed 2020-06-27 by Mr-HaleYa