Mr-HaleYa/MWS-Hardware #46: visual fixes

Mr-HaleYa, 2020-06-27:
swap places
[image]
change layout to this
[image]
swap places
[image]
swap places
[image]
swap places
[image]
add mm labels to these, you can put it in the box or next to the name like "distance to ground in mm)
[image]

0xCTF, 2020-06-27:
**Fixed**

Mr-HaleYa, 2020-06-27:
looks good except for the mm, ill open a new issue for that bc It might be a little work

closed 2020-06-27 by Mr-HaleYa