Mr-HaleYa/MWS-Hardware #48: Station settings visual improvements

Mr-HaleYa, 2020-06-28:
Basically remove the mm label that you added inside the box and put a hover pop-up outside the next to the name in the information column. 

https://drive.google.com/file/d/1NhyyehFrqprL3Wr_2-e0J9BtXY8cAd3r/view

for weir type add (0-no weir 1-rectangular Weir 2-trapezoidal Weir 3-90 degrees v-notch weir)

Mr-HaleYa, 2020-06-28:
might help https://www.w3schools.com/css/css_tooltip.asp

0xCTF, 2020-06-29:
**Fixed**

Mr-HaleYa, 2020-06-30:
@0xCTF 
lol, you forgot the weir type... 

> for weir type add (0-no weir 1-rectangular Weir 2-trapezoidal Weir 3-90 degrees v-notch weir)

0xCTF, 2020-06-30:
> @0xCTF
> lol, you forgot the weir type...
> 
> > for weir type add (0-no weir 1-rectangular Weir 2-trapezoidal Weir 3-90 degrees v-notch weir)

This phrase shown when mouse hover the ❔ ?

Mr-HaleYa, 2020-07-01:
yes, just like you have it here
[image]

Mr-HaleYa, 2020-07-02:
love it ❤️

closed 2020-07-02 by Mr-HaleYa