Mr-HaleYa/MWS-Hardware #51: Data updates

Mr-HaleYa, 2020-06-30:
For now, it would be easier to add a button that says "Refresh"
[image]

0xCTF, 2020-07-03:
Last readings is live updated, but the time&date still not live updated

Mr-HaleYa, 2020-07-03:
live updates are not working for me and the time and date also need to be fixed before closing. It does not update if I have a station page open until I refresh the page. I also discovered a bug when a station has weir == 0 the lastreading box does weird stuff after 5 sec. 

on load 
[image]
after 5 sec
[image]

0xCTF, 2020-07-03:
**Fixed the weird thing when weir_type==0, but still need some work on date&time**

Mr-HaleYa, 2020-07-05:
It works if you change old data manually but it does not show new data being added to the database...

Mr-HaleYa, 2020-07-08:
Some requirements, if you want to do auto data updates
new data is "pulled for" every ~45 sec.
new data is shown on the graph when found
new data is shown on the last reading when found.

we only need this on the STATION page. no other pages need live data

Mr-HaleYa, 2020-07-12:
A refresh button has been added and works great, once live updates are added we will open a new issue.

closed 2020-07-12 by Mr-HaleYa